GameTec-live/ChameleonUltraGUI #27: IOS and MacOS compatability

GameTec-live, 2023-06-12:


retX0, 2023-07-13:
I can help with the Mac/iOS compatible improvement.

GameTec-live, 2023-07-13:
cool, you got a mac and ideally iphone, ideally also know your way around flutter?

retX0, 2023-07-13:
I'm a iOS/Mac developer, so I've the mac and the iPhone, but I'm not so good at flutter,I can learn it, it's just javascript and there has official document in the website. I got it running with my mac, I'm waiting for my ChameleonUltra hardware shipping.

GameTec-live, 2023-07-13:
great, maybe write back when youve recived your hardware...

jphastings, 2023-07-29:
Hi folks, My Ultra has just arrived and I'm keen for an iOS (and optionally a macOS) client too, I want to offer support where I can!

I'm an experienced software engineer, but unfamiliar with iOS & mac development, which means I can build/debug/help test, could maybe contribute to baby-steps PRs (_definitely_ any developer tooling/Ci automation etc.), and have a limited financial budget for helping support anyone's efforts towards making an iOS client.

Please shout if anything there or adjacent would be useful!

GameTec-live, 2023-07-29:
> Hi folks, My Ultra has just arrived and I'm keen for an iOS (and optionally a macOS) client too, I want to offer support where I can!
>
> I'm an experienced software engineer, but unfamiliar with iOS & [...]

hey, thanks for offering your support. As i and basically everyone else currently working on this project do it as a hobby and dont own a mac or have to money to buy one (not even speaking of apples yearly 100€ dev account fee), anyone who wants to help out or has a dev account is welcome to do so. If you just do babystep prs, thats nice too, any help is welcome, especially because i and the other active members (as said) dont own macs and we got enough to do to keep it working on windows, linux and android. If you decide to publish it though, please credit "us" and link this repo. thanks!

beckerj, 2023-08-01:
If anyone else wants to compile/test on mac, here are my steps: 
```
brew install flutter cocoapods
sudo xcode-select --switch /Applications/Xcode.app/Contents/Developer
sudo xcodebuild -runFirstLaunch
git clone https://github.com/GameTec-live/ChameleonUltraGUI.git
cd ChameleonUltraGUI/chameleonultragui
flutter build macos -v
-> This will fail,
    nano macos/Podfile
    -> comment these lines:
#  target 'RunnerTests' do
#    inherit! :search_paths
#  end  
flutter build macos -v
-> This time it should complete successfully
open Runner.xcworkspace (chameleonultragui/macos/) with Xcode and build/run
```

tkerby, 2023-08-07:
This PR fixes the firmware flashing issue: https://github.com/GameTec-live/ChameleonUltraGUI/pull/93

tkerby, 2023-08-11:
A few of us had a look at the MacOS libraries issue preventing nested attacks etc on Tuesday and didn't get as far as we would have liked. 

Looking at the tutorials for ffi, most talk about using the ffi_plugin template which creates a Classes folder in the MacOS directory and appropriate build code for the c libraries to make them .dylib files and call them in the example projects. See https://codelabs.developers.google.com/codelabs/flutter-ffigen#0

Instead, I think when the app was first built it used a different template which has created Xcode projects to build a Mac app at this level. This means incorporating the files into the Xcode projects with appropriate targets to build the libraries correctly. I'm not sure how flutter has managed this and can't find any examples. 

If someone has a little more flutter experience they could perhaps help here. Happy to try any builds on the Mac if you don't have access to one!

thomascannon, 2023-08-18:
I made changes to support building and running on iOS and they are merged into Main now. It is usable but there are some missing things like the nested attacks mentioned above for OSX, as well as some scaling and graphical issues.

Since iOS and OSX are different platforms, wondering if it is best to split this task out.

Foxushka, 2023-08-18:
As mostly only recovery library compilation is left I will rename this issue

pmkid44, 2023-08-19:
> If anyone else wants to compile/test on mac, here are my steps:
>
> ```
> brew install flutter cocoapods
> sudo xcode-select --switch /Applications/Xcode.app/Contents/Developer
> sudo xcodebuild [...]

Works like a charm, thank you!

tkerby, 2023-08-19:
> Works like a charm, thank you!

To fix this issue we can add the runner tests as a new Xcode target

Foxushka, 2023-08-19:
> To fix this issue we can add the runner tests as a new Xcode target

Already removed in 73271e0d882448d1bf81e5174f8a0ec9d3722f99

Foxushka, 2023-08-19:
Pushed first recovery library bindings to https://github.com/GameTec-live/ChameleonUltraGUI/pull/121
iOS next

Foxushka, 2023-08-19:
iOS done too, also there will be compiled artifacts for macOS
Release to App Store soon™

closed 2023-08-20 by Foxushka

komax74, 2023-08-21:
hello thank you very much for the guide, I tried to create the app for macos but it seems to not recognize the chameleon ultra, as if the bluetooth doesn't work, I push in the arrow and A/B button in the CU but they don't show up, there anything to activate first?

Foxushka, 2023-08-21:
BLE is only for iOS, macOS support only cable connection yet

komax74, 2023-08-21:
ok thanks, I also built the iphone app, great, it seems to work fine.
The only thing that doesn't work for me (even with windows and android version) is the initial slider where you should see the occupied slots, it always gives me zero even though i have three active and working both low and high, and this ios version don't slide when push in the arrows.
Is this a known bug or do I have some problem?
Thank you very much.

komax74, 2023-08-21:
probably novice question 🙏 why if I launch the build with the phone attached to the mac everything works well but as soon as I remove the cable the app crashes and don't start again?
Seems it doesn't install it in the phone, is there an option to change to make it stand-alone?

Foxushka, 2023-08-21:
You need to build release build (flutter run ios --release), debug build afaik doesn't contain any code, only Dart VM