GameTec-live/ChameleonUltraGUI #45: Error starting "chameleonultragui.exe" on Windows - file_saver_plugin.dll not found

viperxpi, 2023-07-19:
When starting the app getting and error as can be seen on screenshot. Windows11 Intel machine.

[image]

viperxpi, 2023-07-19:
Issue happens when using the installer. If installed using the installer, but copied the file from the "windows" artifacts - application starts. so it looks like the file was not packed to the installation package.

GameTec-live, 2023-07-19:
what workflow run you used? main branch? (https://github.com/GameTec-live/ChameleonUltraGUI/actions/runs/5596108990)

GameTec-live, 2023-07-19:
hm, indeed, can replicate the issue...

GameTec-live, 2023-07-19:
[chameleonultragui-setup-win.zip](https://github.com/GameTec-live/ChameleonUltraGUI/files/12098774/chameleonultragui-setup-win.zip)
until the fix is merged, heres a manually built installer...

viperxpi, 2023-07-19:
> what workflow run you used? main branch? (https://github.com/GameTec-live/ChameleonUltraGUI/actions/runs/5596108990)

yes. 3 of the latest , reproduced on all of them. but copying the missing file resolves the issue . Only remains to receive the hardware .

GameTec-live, 2023-07-19:
fixed in #46

closed 2023-07-19 by GameTec-live