GameTec-live/ChameleonUltraGUI #116: I would like to add language settings.

jkh0kr, 2023-08-18:
I want to translate it into each country's language.

Is it difficult to add a language file in multi-language without adding the language to the source?

GameTec-live, 2023-08-18:
> I want to translate it into each country's language.
> 
> Is it difficult to add a language file in multi-language without adding the language to the source?

ill look into it and altough it would make sense to have multiple languages, i assume that most people know english, so its going pretty low on the list...

Foxushka, 2023-08-18:
Crowdin integration is worth it, we will accept PR if someone will do it

aruznieto, 2023-08-28:
> Crowdin integration is worth it, we will accept PR if someone will do it

I am trying to do that but It seems that you need premium Crowdin account to do the deliver of the traductions.

GameTec-live, 2023-08-28:
you dont even need crowdin, flutter supports it nativly https://docs.flutter.dev/ui/accessibility-and-localization/internationalization

aruznieto, 2023-08-28:
> you dont even need crowdin, flutter supports it nativly https://docs.flutter.dev/ui/accessibility-and-localization/internationalization

Yes, know. I am doing that, I mention Crowdin because @Foxushka says that

GameTec-live, 2023-08-28:
yeah, relying on a 3rd party isnt the best idea

aruznieto, 2023-08-28:
> yeah, relying on a 3rd party isnt the best idea

👍🏻. [Here](https://github.com/aruznieto/ChameleonUltraGUI/tree/crowdin-integration) you can follow how am I doing it. And [Here](https://crowdin.com/project/chamaleonultragui) is the Crowdin project (only to allow people to translate it, but the integration is manual)

Foxushka, 2023-08-28:
I've already created it 2 days ago: https://crowdin.com/project/chameleon-ultra-gui

Foxushka, 2023-08-28:
> The aren't any string right

I just done some tests, currently busy with macOS/iOS releases. If you need to add some languages or invite you to our project ping me

aruznieto, 2023-08-28:
If you invite me as admin I can manage it

> > The aren't any string right
> 
> I just done some tests, currently busy with macOS/iOS releases. If you need to add some languages or invite you to our project ping me

If you invite me as admin I can manage it

Foxushka, 2023-08-28:
I unfortunately can't add you as member or developer on free account. Should I export and import strings from your project?

aruznieto, 2023-08-28:
> I unfortunately can't add you as member or developer on free account. Should I export and import strings from your project?

I have joined to your project, maybe now you can edit my perms

Foxushka, 2023-08-28:
> > I unfortunately can't add you as member or developer on free account. Should I export and import strings from your project?
> 
> I have joined to your project, maybe now you can edit my perms

"Upgrade to add managers. Managers can create tasks, assign translators, manages glossaries and translation memories, and set up integrations to automate localization." even if I try to add you as developer

aruznieto, 2023-08-28:
> "Upgrade to add managers. Managers can create tasks, assign translators, manages glossaries and translation memories, and set up integrations to automate localization." even if I try to add you as [...]

Ok, when I finish the translation on my project i will send you the files.

aruznieto, 2023-08-29:
> I unfortunately can't add you as member or developer on free account. Should I export and import strings from your project?

I think that is better if you join to my project, because if I need to add a string I will not be able to edit on your project.

Foxushka, 2023-08-29:
To accept translation PR entire app should be translated. You don't need to add new strings after that. If someone will need new strings they could ask in PR with their changes

aruznieto, 2023-08-29:
> To accept translation PR entire app should be translated. You don't need to add new strings after that. If someone will need new strings they could ask in PR with their changes

If you add new strings I need to add it to the translation.  I already translate all texts on app

Foxushka, 2023-08-29:
I still don't understand your point, if I need to add new strings, I add them to our project, then others (and you) can translate them

closed 2023-08-29 by Foxushka