GameTec-live/ChameleonUltraGUI #123: [Bug] Android permission missing for Serial

F9Alejandro, 2023-08-20:
Please see image below for more context.
[image]

[image]

Foxushka, 2023-08-20:
We have different error screen in current release (this belong to web), are you sure you downloaded app from main branch or Google Play?

F9Alejandro, 2023-08-20:
Main branch from gh actions.

Foxushka, 2023-08-20:
Install from google play

Foxushka, 2023-08-20:
And if you want help don't lie next time as I definitely know that this error screen belongs to pending web PR

F9Alejandro, 2023-08-20:
Sorry didn't mean to lie. It isn't like I didn't download the first gh action release that showed up or anything with an apk build....

Foxushka, 2023-08-20:
Enable debug menu in settings and show its screenshot

Foxushka, 2023-08-20:
My bad, no changes there. As I said try to download from google play

closed 2023-08-20 by F9Alejandro