GameTec-live/ChameleonUltraGUI #148: Saving mifare 1k cards to .bin creates empty files

viperxpi, 2023-08-25:
Saving mifare classic 1k cards to .bin creates empty files (0kb, no content), both if saved from the "read card" tab and also when saved from "saved card". Same card saving as .json creates valid json.

Windows version is affected, did not check others

GameTec-live, 2023-08-25:
hm, very interesting, is data on the card? (phone, proxmark 3)

GameTec-live, 2023-08-25:
build number?

viperxpi, 2023-08-25:
> build number?

147.

A correction, only happens when saving from "read file" tab. If saved as card and exported as bin - the bin is not empty.

GameTec-live, 2023-08-25:
@Foxushka looks like your MFC detection pr broke smthng?
because i can replicate, it happens when saving as bin from read page and thats it...