GameTec-live/ChameleonUltraGUI #167: Slot manager shows the wrong card type

Genbox, 2023-08-28:
I have read a card and saved it as can be seen under saved cards:

[image]

However, if I select the Mifare Classic 1K EV1 card (slot 2 in the picture) it shows up as Mifare Classic 2K:

[image]

Should it not say `Mifare Classic 1K EV1` like on the Saved Cards page?

GameTec-live, 2023-08-28:
it shows as MFC2k, because a EV1 contains data on sectors a normal mfc1k doesnt have (16 and 17), but the chameleon only supports 1k, 2k and 4k, so we set the slot to 2k which makes 16 and 17 accesible and then load the card into there. Everything is fine and works as intended, the Slotmanager just shows you a different form of the truth ;p

closed 2023-08-28 by GameTec-live

Genbox, 2023-08-28:
That makes sense. Thanks for the ultra-quick response!