GameTec-live/ChameleonUltraGUI #182: Slot info in slot settings in slot manager

Foxushka, 2023-08-31:
Button at top of slot settings with pencil icon (edit)

Should show UID/SAK/ATQA for HF (missing FW) and UID for LF
Editable ^ data + changing slot type
Download mifare classic dump from slot (save to saved cards), save lf to slot too

Foxushka, 2023-08-31:
No, slot should be editable too

GameTec-live, 2023-08-31:
no

Foxushka, 2023-08-31:
yes

cyber-vi-king, 2023-09-01:
I agree, setting basics per slot are useful (also just viewing the current UOD/SAK/ATQA would help already). 

For some locks there is no need to set a dump if the uid is enough.

GameTec-live, 2023-09-01:
https://github.com/GameTec-live/ChameleonUltraGUI/tree/slot-editing
Waiting for fw

Foxushka, 2023-09-06:
Firmware part finished in https://github.com/RfidResearchGroup/ChameleonUltra/pull/104

PR #334 (2023-10-14): View slot info and editing
Something is done on purpose:
You can't access Mifare Classic emulator settings before loading data to slot
Mifare Classic emulator settings are changed independently from UID (without clicking save button)
Restricted access to NTAG (not supported by firmware, don't let users to break whole app)
Unknown tag type can't be saved (as data wiping done by other button)

Fixed a lot of bugs, code quality improved

Closes #182
closed 2023-10-14 by Foxushka