GameTec-live/ChameleonUltraGUI #185: Firmware update via iOS corrupts device

wasp604, 2023-09-01:


GameTec-live, 2023-09-01:
you can use serial on ios? Anyways, on ios your supposed to use BLE only...

Foxushka, 2023-09-01:
Does progress bar went to end? Doesn't your phone went to sleep while flashing? How much times you tried to flash?

Foxushka, 2023-09-01:
And what build is in Settings - About?

wasp604, 2023-09-01:
Apologies for misunderstanding, the Chameleon was connected via USB-C but updated via BLE.

Build # 1.0.246 (246)
I have attempted to update at least 10 times now, deleting app and retrying, restarting phone and to trying, etc.

wasp604, 2023-09-01:
The phone did not / does not sleep while flashing

wasp604, 2023-09-01:
Note: there is no progress bar visible on the iOS app, only blue flashing LEDs on Chameleon Ultra that stop after some time and result in an unfinished update DFU mode state.

wasp604, 2023-09-01:
iOS app also never shows progress, never "finishes", only LEDs on Chameleon stop blinking.

GameTec-live, 2023-09-01:
This seems to be a IOS only issue, on 246 i can flash fine (android). Someone else on the discord was able to replicate this issue though.

masasron, 2023-09-01:
> Note: there is no progress bar visible on the iOS app, only blue flashing LEDs on Chameleon Ultra that stop after some time and result in an unfinished update DFU mode state.

Same here, flashing from macOS works.

aruznieto, 2023-09-01:
Yes, I try to flash with my iPad connected through BLE and It does not work, and I cannot connected it with the cable Type-C, but It does not break the Chameleon, the Cham enter on DFU mode.

TXSafeCracker, 2023-09-01:
Same here, getting corrupted firmware when updating on iOS via BLE

Foxushka, 2023-09-01:
Build 250 will have debug logger into debug settings (waiting for apple review). Click `Enable production logs`, restart app (just to be 100% sure), flash via DFU, wait chumleon to go to sleep, exit app, go to debug settings and click `Copy logs to clipboard`. Then upload to pastebin or something else

auroraviii, 2023-09-03:
I have the same issues.   It's found. Says it's now stuck in DFU, it proceeds to flash. First rapid green on 4-5 and then blue.    

There is zero progress bar or indicator it worked.    It just stops during blinking blue on 4-5.    

Then only button A will work, then the app tells you you're stuck in DFU.

Around and around it goes.

auroraviii, 2023-09-03:
> > Note: there is no progress bar visible on the iOS app, only blue flashing LEDs on Chameleon Ultra that stop after some time and result in an unfinished update DFU mode state.
> 
> 
> 
> Same here, flashing from macOS works.

I'll try from macOS instead of iOS! Hopefully that's the problem

GameTec-live, 2023-09-03:
> > > Note: there is no progress bar visible on the iOS app, only blue flashing LEDs on Chameleon Ultra that stop after some time and result in an unfinished update DFU mode state.
> > 
> > 
> > Same here, flashing from macOS works.
> 
> I'll try from macOS instead of iOS! Hopefully that's the problem

if you could send the logs it would help, macOS doesnt use bluetooth so its a completly different system...

auroraviii, 2023-09-03:
I have turned on debug mode.  Forgive me but where can I locate the logs in debug mode?   I would like to send 

I'm extremely good with the flipper zero.  But I'm feeling lost 

It's just a ton of warnings and functionless buttons.

lumcsyahoo, 2023-09-03:
Yes, the IOS version on the iphone corrupts the firmware on the Ultra, to the extend that the bluetooth function is affected and could not connect on the Mac. You got to use a USB-C cable to do the connection and reload the firmware, then it works fine.

auroraviii, 2023-09-03:
Ok I will update with USB C at this time.    Thank you

I will locate the debug logs to post...

GameTec-live, 2023-09-03:
if on the latest version you should just be able to scroll down

auroraviii, 2023-09-03:
> if on the latest version you should just be able to scroll down

I just downloaded it.  It's current. Will not scroll, just flip. I am on iOS 17.0... 

No worries. I'll grab it later on different device.  Thank you

Foxushka, 2023-09-03:
No apple approved version has logging into it. Wait until apple will review updates

auroraviii, 2023-09-03:
> No apple approved version has logging into it. Wait until apple will review updates

Oh my mistake.  Thank you 😅

Foxushka, 2023-09-05:
If someone want to help it fix it faster: share your email and I will invite you to internal testing, so we don't need to wait for apple to approve app

auroraviii, 2023-09-05:
Still the same for now.   

Pro: Scrolling is fixed under debug menu.   

Con:
1. Still unable to pull logs as none of the item to click on work. They just turn blue.

2. Still stops with DFU screen up and 4-5 flashing blue then stops just stops.  

Iam on test flight as requested. I have sent feedback to Andrew as well. 

[image]

Foxushka, 2023-09-05:
Upload copied logs into pastebin and send link there

auroraviii, 2023-09-05:
Yes understand. Unfortunately when I retry DFU process (fails). Using with TestFlight...enable debug...press copy logs to clipboard.   Nothing will show up/or paste into ANY including pastebin. 

Non of the debug links work.

If I'm missing something obvious I'm sorry!

Open collective donated for iOS!

GameTec-live, 2023-09-05:
> Probably logs are broken for mobile devices, haven't tried them. Will fix later (or @GameTec-live, probably android and iOS issue is same)

android issue doesnt exist, its a bonding messup

Foxushka, 2023-09-05:
For me logs aren't copying on android, do they for you?

auroraviii, 2023-09-05:
Negative

Foxushka, 2023-09-06:
Okay, build 281 pushed to TestFlight, should have both logs and DFU flashing fixed. Waiting someone else to try and report on that (if flashing doesn't work, enable logs in debug menu and try flashing again, then upload logs to pastebin and send there)

auroraviii, 2023-09-06:
No go.   Now it does recognize the chameleon faster when 4-5 are flashing red and DFU mode.

It does not show a new Downloading screen while 4-5 blink red.   

Then nothing.

4-5 will remain fashing but is not shown.

Anle to paste logs on new test flight build.  

#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 0
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!

Foxushka, 2023-09-06:
Interesting, you tried flashing from connect menu or from settings?

auroraviii, 2023-09-06:
I booted up the app once update in test flight.   Pressed B.  4-5 flashing red.  Pressed refresh, it was found.  Then on the bottom it says ("downloading firmware VERY quickly" then it's over.  

No green>blue 4-5
No longer do you see the main "DFU reflashing" screen.   Just blank

4-5 will continue to blink then stop.  

Most recent:

#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 0
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 0
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
[... 28 lines cut ...]
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
[... cut ...]
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!

Foxushka, 2023-09-06:
Can you flash chumleon from other device and try flashing from settings?

auroraviii, 2023-09-06:
I can try from windows around 6pm est NY

I just tried from debug settings menu with iOS.  The button for "force flash ultra firmware " does not seem to do anything.    Yes it's flashing 4-5 red when I try it

Foxushka, 2023-09-06:
I mean chameleon settings, not debug menu. You can't access chameleon settings without working firmware. Anyway, I will check flashing from main menu later

auroraviii, 2023-09-06:
OK so I tried the normal safe installation process of updating firmware on an iPhone SE 2nd est with firmware 16.6.

It had a progress bar on bottom and it WORKS!!   But I can't pull logs lol

I was on my iPhone 12 Pro Max with 17.0 beta 7. It's quite possible that somehow this is an issue for the future iOS of 17.   

Regardless there is a difference.   16.6 works.   17.0 beta 7 does not!

It's working now on my iOS 17.0!

I would say it's fixed!

[image]

auroraviii, 2023-09-06:
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 0
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 0
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.availableDevices (package:chameleonultragui/connector/serial_ble.dart:34)[0m
[38;5;208m#1   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:83)[0m
[38;5;208m⚠️ Multiple searches in one time not allowed! FIXME[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:73)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:102)
[... 571 lines cut ...]
🐛 Received message: command = 1026, status = 104, data = 01
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:365)
#1   ChameleonCommunicator.getButtonConfig (package:chameleonultragui/bridge/chameleon.dart:852)
🐛 Sending: 11ef040200000001f942be
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:337)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1026, status = 104, data = 02
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:365)
[... cut ...]
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:365)
#1   ChameleonCommunicator.getBLEConnectionKey (package:chameleonultragui/bridge/chameleon.dart:894)
🐛 Sending: 11ef040700000000f500
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:337)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1031, status = 104, data = 313233343536

Foxushka, 2023-09-07:
I won't call it fixed until I will fix flashing from connect page

Foxushka, 2023-09-07:
Should be fixed in #235

Foxushka, 2023-09-08:
Build 289 now in TestFlight, @auroraviii can you retest flashing from main menu? (You can click enter DFU in chameleon settings)

auroraviii, 2023-09-08:
Yes it works.  DFU activated from settings once TestFlight updated.  

Ran DFU>firmware update 100%

No issues !

Foxushka, 2023-09-08:
Enable DFU -> flash from connection menu? If yes, try to flash 3 times in a row and I will assume it is fixed completely

auroraviii, 2023-09-08:
Ok I attempted 3 times. DFU>reflash. 

Currently
"Firmware up to date"

Looks to be working in my world

See logs

#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getUsedSlots (package:chameleonultragui/bridge/chameleon.dart:756)
🐛 Sending: 11ef03fb000000000200
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1019, status = 104, data = 04010300000100000000000000000000
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getEnabledSlots (package:chameleonultragui/bridge/chameleon.dart:840)
🐛 Sending: 11ef03ff00000000fe00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1023, status = 104, data = 0101010100000000
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0002fe
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0001ff
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0102fd
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0101fe
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0202fc
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0201fd
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0302fb
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
#1   ChameleonCommunicator.getSlotTagName (package:chameleonultragui/bridge/chameleon.dart:704)
🐛 Sending: 11ef03f0000000020b0301fc
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:347)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1008, status = 113, data = 
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:375)
[... 3524 lines cut ...]
#0   DFUCommunicator.sendCmd (package:chameleonultragui/bridge/dfu.dart:171)
[... cut ...]
#0   DFUCommunicator.sendCmd (package:chameleonultragui/bridge/dfu.dart:180)
#1   <asynchronous suspension>
🐛 Received: 600101
#0   DFUCommunicator.sendFirmware (package:chameleonultragui/bridge/dfu.dart:290)
#1   DFUCommunicator.flashFirmware (package:chameleonultragui/bridge/dfu.dart:263)
🐛 Serial: Streaming Data: len:4096 offset:180224 crc:0x51b65097 mtu:2051
#0   DFUCommunicator.sendCmd (package:chameleonultragui/bridge/dfu.dart:171)
#1   <asynchronous suspension>
🐛 Sending: 03
#0   DFUCommunicator

Foxushka, 2023-09-08:
Good to know, thanks for fast testing

GameTec-live, 2023-09-08:
accorsing to 2 people alrdy, its fixed on build 286. Anyone who can confirm or deny that?

auroraviii, 2023-09-08:
There is a lot of posting on YouTube as of 8 hours ago...Of it working, no complaints!

   Since no one has responded here.   It least we can see something

closed 2023-09-08 by GameTec-live

wasp604, 2023-09-20:
iOS build 317, unable to check keys against user dictionary, firmware repeatedly corrupts just like before, except this time it has a progress bar and stops halfway.

wasp604, 2023-09-20:
And of course, after 5 times failed flash, enabling debug mode with production logging allows it to complete flash.

Sorry for the inconvenience.