GameTec-live/ChameleonUltraGUI #188: Support a more compact UI on small / high density screens

bwachter, 2023-09-01:
Attached is a screenshot from a 6" screen, with only a small part of the card for one technology visible. The space should be sufficient to keep a complete card - including the key options - visible here.

[image]

bwachter, 2023-09-01:
Scrolling (at least on non touch screen devices) is quite annoying for this - and keyboard navigation also isn't behaving very well (I've figured out earlier that I can move between elements with arrow keys, and if I'm on the card side pgup/pgdown are starting to work - which is neither fast nor intuitive). The application looks promising for quick actions on the road on small devices - but the layout gets in the way.

I've been looking into flutter for setting up proper RPM builds of this few days ago, and wasn't very impressed about the overall state of this if you don't just want to grab the binary packages from google - so I'm unlikely to do code contributions

GameTec-live, 2023-09-01:
Oh and because of your flutte issue, i think you can build from source...
https://github.com/flutter/flutter/wiki/Compiling-the-engine

bwachter, 2023-09-01:
> why are you using a gui without a touchscreen or mouse? lol

I do have a touchpad on that thing (so I can click stuff), but two finger scrolling is working badly - and even worse when I'm outside and might have dirty fingers. Device is a GDP Micro PC, which generally is great as a portable machine with more options than an Android device.

> And your the first guy where "the layout gets in the way"... its a standard layout with a navbar on the side... I know the app doesnt scale perfectly, but its a pretty standard layout that works (imo) pretty well.

This (and the chameleon ultra) generally seems to be in early stages of development, so I'm not surprised about that :) Apart from cracking mifare classic keys so far the proxmark seems to be more useful, though development looks promising.

A compact UI would have an advantage in speed for some operations over CLI - though something properly scriptable like the proxmark client isn't far behind.

GameTec-live, 2023-09-01:
well, were still using material design and flutter (which is widgeting layouting), so compact UI isnt gonna happen... IDK what your gonna do about your scrolling "issues" then, tbh sounds more like a problem your gotta solve, because i imagine a lot of other apps will behave similarly... As this App is available on andoird too and offers the exact same feature set (even more, BLE!), id suggest switching over to android or using the app in portrait on your gdp :p

GameTec-live, 2023-09-01:
thx for the feedback and i know that scaling has to be looked at again, especially for "special cases" like you got, but thats currently one of our least concerns... anything else or am i safe to close this issue for now?

Foxushka, 2023-09-01:
https://pub.dev/packages/flutter_screenutil

Akisame-AI, 2023-09-04:
I don't mind implementing a scaling option in the settings menu that allows for custom scaling. 
The package from Foxushka could work but it is also still under development.

Foxushka, 2023-10-18:
Is there any other flutter apps that support such behavior?

djvnsk, 2026-03-08:
Hi, I'd like to add a couple of interface improvement requests.
1. In the card reading section, it would be great to stack the buttons, because on a narrow screen it's impossible to even see what's written on them.
2. In the saved cards section, the card and dictionary lists are scrolled separately, not as a single list. With large paddings, it's difficult to manage a list with many items. The open and add buttons could be moved next to the section header, and in some dictionary cards, the buttons shift to the next line.
3. There's an inconvenience that complicates localization (translation) of the app: the "row" names don't indicate the context (section or screen) in which the row is located. I think it would be more convenient to see something like "home_device_settings_animation_full" instead of "full".
[image]
[image]