GameTec-live/ChameleonUltraGUI #205: Editable Hex

Markg546, 2023-09-03:
After copying a card it could be useful to have a built in hex editor to change some data on the move (without having to dump and then edit externally and then import again)

GameTec-live, 2023-09-03:
it is on the todolist, but good hex editors are a pain... afaik @thomascannon is working on one

sarah1331sarah, 2023-09-03:
@GameTec-live i seem to remember you mention that edit dump option was "to do" did you ever consider appenhend keys from dump to file? or keys found to key file?

GameTec-live, 2023-09-03:
import, export and editing uid, etc, is all working, only hexeditor is missing. On the roadmap (in order of prob going to be implemented):
Better dict management (merging dicts mostly)
Compatability with proxmark json
Write card page
Hex editor

telephasicworkshop, 2023-11-28:
If editor is difficult or impractical, suggestion to display the full hex in a simple viewer in the meantime?

GameTec-live, 2023-11-29:
> If editor is difficult or impractical, suggestion to display the full hex in a simple viewer in the meantime?

There is a draft, with a very very janky hex editor... @augustozanellato Wanted to redo it properly and modular (which im thankful for) and hes almost done, eycept he cant figure out a good way to do input, which is kinda essential for a hex editor...

sarah1331sarah, 2023-11-29:
@GameTec-live anywhere with the write function I hate using the MTools app to write with the ultra

telephasicworkshop, 2023-11-29:
Nice to hear! Will stay tuned.
Even the ability to just view the hex would add functionality and usefulness for analysis.

augustozanellato, 2023-11-29:
> Nice to hear! Will stay tuned. Even the ability to just view the hex would add functionality and usefulness for analysis.

I already have hex view working, what’s bothering me is the editing/input. I might cleanup my code shitshow tomorrow and push it here.

Sarah-1331, 2024-04-01:
@augustozanellato did you manage to get the hex editor working to any degree? apart from the view i think view would still be a massive help buddy