GameTec-live/ChameleonUltraGUI #214: Some appearence issues on Android smartphone

spp2000, 2023-09-04:
Hi, all the pages looks fine now, but there are 4 that need some improvement in my case. I show you:

**Text size is much bigger than the others (bigger than title)**

[image]

**Truncated buttons**

[image]

**Truncated text** (the slot nick can be even longer than this)

[image]

**Truncated text and sectors**

[image]

Thank you

Foxushka, 2023-09-04:
Should I move this to https://github.com/GameTec-live/ChameleonUltraGUI/issues/188? Maybe. Issues are known and should be fixed sometime. Current priority is to implement everything important, and only then move on to UI fixes.

spp2000, 2023-09-07:
Hi, just to inform you that truncated button in Slot Settings (2nd picture) now are ok! Thanks

spp2000, 2023-09-09:
"Truncated text and sectors" (4th picture) both solved! Nice layout.

But, after checking keys with some unknown, "Dump partial data" button is partially truncated

GameTec-live, 2023-09-09:
Its because of this fixed spacer: https://github.com/GameTec-live/ChameleonUltraGUI/blob/de2513cb75a0e7b9d7606e4bb71073d07dc088da/chameleonultragui/lib/gui/page/read_card.dart#L808

Foxushka, 2023-09-10:
@spp2000 should be fine on build 305+. Other still depends on #200

spp2000, 2023-09-11:
> @spp2000 should be fine on build 305+. Other still depends on #200

Yes, now the two buttons are complete and well centered!! Thanks
Also the 1st picture (huge text) depends on #200?

Akisame-AI, 2023-09-17:
The saved cards page seems to break when I add in my scaling fixes to it. Saved cards should have a lot smaller letter type already though. 
I will try scaling in a different way and submit it. I have a bit of free time today.

Akisame-AI, 2023-09-17:
Okay, scaling issues should be a lot better now. Not prefect but at least usable. 
These scaling things only apply on very narrow windows. 
[image]

[image]

[image]

[image]

PR #215 (2023-09-05): BLE settings, GUI overflow fixes, other fixes...
Closes #213, partial fixes #214 (in chameleon settings only)
closed 2023-10-09 by GameTec-live