GameTec-live/ChameleonUltraGUI #218: Recover keys (mfkey32) timeout on iOS

krissen, 2023-09-05:
Recovering keys through collected nonces works fine on Android. 

When attempting on iOS, recovery timeouts `waiting for response for command 4005`. 

[image]

Connected through secure BLE. Connection good, everything else works fine. Only key recovery timeouts, and only on iOS.

krissen, 2023-09-05:
Hold up. Might not be particular to GUI.

In cli, `hf detection count` (for instance) also gives me the same error:

```
TimeoutError: CMD 4005 exec timeout
```

Over serial. Nevertheless, on Android (BLE, like iOS) *it works.*

GameTec-live, 2023-09-05:
> Hold up. Might not be particular to GUI.
> 
> In cli, `hf detection count` (for instance) also gives me the same error:
> 
> ```
> TimeoutError: CMD 4005 exec timeout
> ```
> 
> Over serial. Nevertheless, on Android (BLE, like iOS) _it works._

maybe talk to the firmware devs then?

krissen, 2023-09-05:
> maybe talk to the firmware devs then?

Naturally. Hadn’t checked with cli prior to submitting the issue, only iOS and Android. Given that one worked and the other didn’t, a GUI issue seemed more plausible. 

Now it doesn’t. 

Let’s close the issue for now at least.

closed 2023-09-05 by krissen