GameTec-live/ChameleonUltraGUI #224: Changing the default Bluetooth pin breaks the connection to the device

georgikoemdzhiev, 2023-09-06:
Hello, 

I tried changing the default Bluetooth pin (12345) via the app but after pressing save, I am no longer able to pair with the ChamelionUltra. It keeps prompting me to type in the password. When I type in the correct, new password nothing the device icon appears in the "Connect" screen but when I click on the device's icon I am prompted to type the password again (and that repeats endlessly).

GameTec-live, 2023-09-06:
you probably need to delete the bonds / unpair? https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/troubleshooting.md#ble

Foxushka, 2023-09-06:
[image]
🙄

Foxushka, 2023-09-06:
[image]
Required on change password too... Eh.

Foxushka, 2023-09-06:
[image]

closed 2023-09-06 by Foxushka