GameTec-live/ChameleonUltraGUI #237: LF tag uid 01?

Barishhh, 2023-09-07:
When reading an LF tag, I can save and assign this to a slot. 
When exporting the LF data to .json, I can see the tag uid is 01. Is this working as expected? Might be a noobie question, but shouldn't this be the id of the tag instead? Is this either a bug or am I doing something wrong?

Foxushka, 2023-09-08:
Can you show example? It's working fine for me

Barishhh, 2023-09-08:
Sure, contents of the .json file:

{"id":"c55f0f20-9234-44bd-a6e2-41f94df914e6","uid":"01","sak":0,"atqa":[],"name":"Card","tag":1,"data":[]}

Foxushka, 2023-09-08:
Can you try to dump card again, will it be 01 again?

Barishhh, 2023-09-08:
Updated the GUI, now my tag doesn't get recognized at all. 
Starting to wonder if this is even a LF card. I thought it is. 

It looks like the card on the right on this page: https://bel-cebeo.mo.cloudinary.net/PRODUCT/DOCUMENT/PictureServer/default.aspx?947FB0959B848C849988A89AA47C8390857A8476836E8C6881768D7C886C5EA4B4AB&ts=1683520685367

Foxushka, 2023-09-08:
This is mifare classic (HF) card, not LF. Idk how it got recognized via LF, we blindly trust what chameleon reply to us. So 01 is fluke, not a bug.