GameTec-live/ChameleonUltraGUI #266: Actual release versions and change logs .

trivarialthea, 2023-09-14:
**Is your feature request related to a problem? Please describe.**
No

Would it be possible to include an actual changelog for the firmware and app before and after updating somewhere? 
As currently the AppStore changelog does not tell much, apart from what pull request was merged and in terms of firmware there is nothing at all.

**Possible solutions**
1. Show everything in app
2. Show the app changelog in respective app stores or GitHub release page
3. Redirect to a changelog website from within respective sources? (Chameleon ultra website etc.)

**Notes on my request**
I do understand that for the app I can just lookup the pull request and read it from there, but for an BFU this is a complicated process where that user would probably not even know where to look. So it would be a "nice to have" in the future.

Thanks for understanding and reading this request, feel free to comment or criticize it up ever you want, I will try to answer as much follow up questions I can 😉.

GameTec-live, 2023-09-14:
might be useful, but its extra work, especially since currently most be bugfixes and when smthng major gets merged itll be in the pr merged title anyways... Github releases will be published sooner than later....

trivarialthea, 2023-09-14:
> might be useful, but its extra work, especially since currently most be bugfixes and when smthng major gets merged itll be in the pr merged title anyways... Github releases will be published sooner than later....

Makes complete sense, the easiest version I can see, if releases are a thing, is requesting the release description via the GitHub api and making the changelog be available in app that way. 
Same could be done for the firmware if RRG would start using them too 🤔

Foxushka, 2023-09-17:
Actually, not possible with at least firmware. It's changelog doesn't have any per commit info. App... I don't like when app shows that it was updated and changes. We don't have any OTA on any platforms (not app based). Where we have automatic updates, we have last commit message. 

From the other side, it's possible to implement changelog somewhere in app settings and put it on build time. But requires a lot of work, as build number goes from github actions... And each time we need to parse all actions to get commits, messages and etc.

GameTec-live, 2023-09-17:
indeed.... and i dont think that most people are interested in a detailed changelog, and if they are, they can visit the github... Most people dont read the changelog and just go "huh, thats a cool new feature" or "huh, that bug got fixed, nice"...

trivarialthea, 2023-09-20:
Fair point in terms of the arguments. 
I think it would still be beneficial to have at least some general information about the updates, but if that is not possible, due to it going on a "per commit" basis and no release cycle is planned, it makes sense.
Though if a release cycle is eventually implemented for the FW and/or for the app, it could be a decent feature. I also fully agree with @Foxushka that it is a low-priority suggestion/Feature Request.