GameTec-live/ChameleonUltraGUI #267: Mifare Classic 1K recover key problem

alonegr, 2023-09-14:
Hi everyone, with both the Android and Windows apps updated to the latest version available, it is impossible for me to recover the keys for a normal Mifare Classic 1k while it works if I use a copy on a magic card cloned with Promark the Chameleon Ultra is updated to the latest version. Thank you
[image]

GameTec-live, 2023-09-15:
what is the output of the pm3s HF search? because if the card uses hardnested it isnt supporred yet.

Foxushka, 2023-09-15:
Staticnested/hardnested not supposed on firmware side. Nothing we can do. Create issue on Chameleon Ultra firmware repo

closed 2023-09-15 by Foxushka