GameTec-live/ChameleonUltraGUI #270: Checking user dictionary keys no longer shows animation / no longer working

wasp604, 2023-09-20:
When checking read HF card against user dictionary, loading animation no longer shows. 

**To Reproduce**
1. Read HF card 
2. Select a user dictionary (I'm using flipper default)
3. Button shows pressed, but nothing happens.

**Expected behavior**
Loading animations after "check keys from dictionary" button is pressed.

**Screenshots**
[image]

**Version info:**
 - OS:iOS 16.6.1
 - App build number:317
 - Connection type:BLE
 - Chameleon firmware commit:98605be

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [x] - I've searched and verified that there is no already created issues 

**Additional context**
Add any other context about the problem here.

wasp604, 2023-09-20:
CAUSE FOUND:
Card is password-protected Mifare Ultralight 21, according to Flipper Zero reader.

Animation shows up with non-password-locked cards

GameTec-live, 2023-09-20:
password protected ultralight, does the chameleln even support ultralights?

Foxushka, 2023-09-20:
Very interesting that it shows key checkmarks, should not.

wasp604, 2023-09-20:
> password protected ultralight, does the chameleln even support ultralights?
> 
> But interesting that thats a error or even a thing lol... i do need to get some more cards...

[Official documentation](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/technical_whitepaper.md) says it's supported but not yet implemented.

Foxushka, 2023-09-20:
1. Never trust those claims, Ultralight/NTAG isn't supported yet
2. Those are only for Mifare Classic, Mifare Ultralight has only one password

wasp604, 2023-09-20:
> 1. Never trust those claims, Ultralight/NTAG isn't supported yet
> 
> 2. Those are only for Mifare Classic, Mifare Ultralight has only one password 

Not sure what you mean there, since the official documentation agrees with what you and I said, but your wording is that of disagreement.

It says in the chart the hardware supports it but that it isn't implemented yet. I was just using the terminology of the documentation page.

Foxushka, 2023-09-20:
> Not sure what you mean there, since the official documentation agrees with what you and I said, but your wording is that of disagreement.

For us if it isn't available on firmware side it isn't supported. For firmware side it isn't implemented

PR #272 (2023-09-22): Don't rely only on ATQA/SAK while getting card type
Closes #270
closed 2023-09-22 by Foxushka