GameTec-live/ChameleonUltraGUI #275: Improve key finding in dictionary

maul3x, 2023-09-22:
**Describe the bug**
When I try to read a card with a big dictionnary (1600keys), it's very slow. But the key is the same between each sector, it would be fast after the first key was find ?

GameTec-live, 2023-09-22:
> it would be fast after the first key was find ?

what?
Im just gonna assume you mean that one should cache the first found key and try it on the next sector, of that fails start again from dict. if thats what you mean itd make sense, yeah, not a bad idea

maul3x, 2023-09-23:
YES ! It's exactly what I mean.
(Sorry for my english)

Foxushka, 2023-09-23:
This adds additional level of complexity to key checking. I don't sure we are ready for it

Foxushka, 2023-10-13:
Will try to fit in 1.0.0 release

PR #331 (2023-10-13): Recheck found keys on other sectors
Closes #275
closed 2023-10-13 by Foxushka