GameTec-live/ChameleonUltraGUI #289: Check for new version on startup.

vaanja, 2023-09-28:
**Is your feature request related to a problem? Please describe.**
Currently the application does not check the availability of the newer version in any way, if the user wants the latest version, he has to download and install the installation package from the repository.

**Describe the solution you'd like**
Each time you run the application, it checks for the availability of a new version and offers an update if a newer version exists.

**Describe alternatives you've considered**
If automatic checking for new version at startup would not be implemented, at least add checking for new version in Settings-About.

Foxushka, 2023-09-28:
OTA is very hard for cross-platform and I don't think we are planning it. Automatic update check also won't be available (for privacy). And we don't yet have versioned releases anyway

GameTec-live, 2023-09-28:
+ I personally find apps bugging me to update annoying AF, so im not gonna implement it... 

God im gonna update my apps when I want to not when the app wants to!