GameTec-live/ChameleonUltraGUI #291: Timeout errors

auroraviii, 2023-09-29:
**Describe the bug**
Once all devices were updated to the most current TestFlight, software and firmware (DEV4). I will get constant timeout errors, 1003 and 4005.

Also, one retrieving keys in any slot, this will happen instantly.   

**To Reproduce**
Steps to reproduce the behavior: Recovering keys in any slot=4005

**Expected behavior**

Since update and DFU update, I will constantly get timeout errors.  

**Screenshots**
[image]

**Version info:**
 - OS: 17.1, iPhone 12 Pro Max 

 - App build number: 1.0.331

**Also required**:

- [ ✅] - I've upgraded app to latest app version

- [ ✅] - I've upgraded Chameleon firmware to latest version

- [ ✅] - I've searched and verified that there is no already created issues 

**Additional context**

#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1023, status = 104, data = 01010100000101010000000000000000
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1EmulatorConfig (package:chameleonultragui/bridge/chameleon.dart:819)
🐛 Sending: 11ef0fa9000000004800
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 1023, status = 104, data = 01010100000101010000000000000000
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1EmulatorConfig (package:chameleonultragui/bridge/chameleon.dart:819)
🐛 Sending: 11ef0fa9000000004800
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4009, status = 104, data = 0100010100
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1DetectionCount (package:chameleonultragui/bridge/chameleon.dart:652)
🐛 Sending: 11ef0fa5000000004c00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4009, status = 104, data = 0100010100
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1DetectionCount (package:chameleonultragui/bridge/chameleon.dart:652)
🐛 Sending: 11ef0fa5000000004c00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4009, status = 104, data = 0100010100
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1DetectionCount (package:chameleonultragui/bridge/chameleon.dart:652)
🐛 Sending: 11ef0fa5000000004c00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4005, status = 104, data = 00000012
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4005, status = 104, data = 00000012
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.isMf1DetectionMode (package:chameleonultragui/bridge/chameleon.dart:641)
🐛 Sending: 11ef0fa7000000004a00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4007, status = 104, data = 01
#0   ChameleonCommunicator.sendCmd (package:chameleonultragui/bridge/chameleon.dart:381)
#1   ChameleonCommunicator.getMf1DetectionCount (package:chameleonultragui/bridge/chameleon.dart:652)
🐛 Sending: 11ef0fa5000000004c00
#0   ChameleonCommunicator.onSerialMessage (package:chameleonultragui/bridge/chameleon.dart:353)
#1   BLESerial.connectSpecificInternal.<anonymous closure>.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:202)
🐛 Received message: command = 4005, status = 104, data = 00000012

Foxushka, 2023-09-29:
Doesn't seem to reproduce for me. Can you record video what are you trying to do?

auroraviii, 2023-09-29:
https://github.com/GameTec-live/ChameleonUltraGUI/assets/70302497/54bf83cc-39dc-4e4e-80f8-7c5b94a7a779

Foxushka, 2023-09-29:
Interesting bug, I've reproduced it. Data actually comes into app and it correctly parses it, but somehow can't pick it up... No ideas how to fix it yet

auroraviii, 2023-09-29:
It's not consistent. But it's definitely holds for a while. I have to do a full phone reboot. Full reflash, which is weird because it technically should have already the most current firmware but it says it does not.  And then that issue will proceed.

Foxushka, 2023-09-29:
I know, no need of doing anything from your side

PR #296 (2023-10-03): Try to resend command if timeout
Closes #291
closed 2023-10-03 by Foxushka