GameTec-live/ChameleonUltraGUI #320: Add support for more TAG types

bagbagbag91, 2023-10-11:
I'm using Chameleon Ultra for a few weeks. I'm very happy that we already have such a good GUI. 
So far I've managed to successfully clone `EM410X` and `Mifare Classic 1K` tag types and It worked lie a charm.

But recently I took a TAG from my friend and while trying to clone it here is what I've got:

[image]

Would be really great if you add support for more TAG types.

I know that Chameleon Ultra supports a lot of HF and LF TAG types, [here](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/technical_whitepaper.md#high-frequency-reader) is a full list of supported TAG types.

GameTec-live, 2023-10-11:
it doesnt support "a lot of tag types", if it doesnt say "supported" in all 3 collumns, we cant implement it. And if your talking bout ntag, #278

closed 2023-10-11 by GameTec-live

bagbagbag91, 2023-10-11:
I see, thank you 🙏