GameTec-live/ChameleonUltraGUI #345: App triggers from device buttons

tkerby, 2023-10-16:
Would be incredible to be able to select a slot then long press to trigger an action in the app such as reading, running a nested attack and writing back to the slot. Would allow for covert operations. 

May need to set the app to stay open in some way to monitor button presses

GameTec-live, 2023-10-17:
> Would be incredible to be able to select a slot then long press to trigger an action in the app such as reading, running a nested attack and writing back to the slot. Would allow for covert operations. 
> 
> May need to set the app to stay open in some way to monitor button presses

the app staying open wouldnt be the problem, just dont let your phone sleep... the bigger problem would be, that wed somehow need to monitor the buttons, etc. Thatd need to be a fw thing and idk how useful thatd be, as the cli wouldnt have much of a use out of it and you need to have a running client connected...
Wed also prpbably need a "priority" system, so that the button action happens in a timly manner...

Foxushka, 2023-10-17:
No, why? If you need to open app, you could afford clicking 2 more buttons to read card :roll_eyes:

GameTec-live, 2023-10-17:
why not, its a interesting concept... And as said, that allow for more covert operation, just having your phone tucked away and being able to run a full clone for example would be useful for some... its very low priority, but why not...

Foxushka, 2023-10-17:
Then move issue into firmware repo, we can't (and won't) do anything with that. This requires huge protocol modification

Foxushka, 2023-10-17:
As I said, not possible without firmware. Create issue in firmware.

closed 2023-10-17 by Foxushka

GameTec-live, 2023-10-17:
ok, then @tkerby please open a issue for the firmware side (sending button presses to the client) and this will be reponened when thats implemented 🤷