GameTec-live/ChameleonUltraGUI #367: MiFare 4K

aruznieto, 2023-10-29:
**Describe the bug**
Month ago (more or less) I could to read a dump a MiFare 4K card. Today I can't do it. The button of dump not appear

**Version info:**
 - OS: iOS 17
 - App build number: last (BETA)
 - Connection type: BLE
 - Chameleon firmware commit: v2.0.0-114-gec3c06

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [x] - I've searched and verified that there is no already created issues

Foxushka, 2023-10-29:
Record video

GameTec-live, 2023-10-29:
tried on my iPad with a 4k Card, it works fine....
iOS: 17.0.3
App: 375
FW: 2.0.0-104-g299562b

aruznieto, 2023-10-29:
In my MAC, iPhone and Windows it does not work. The technology is "Other"

GameTec-live, 2023-10-29:
lemme try on windows, but are you sure its a MFC 4k?

Foxushka, 2023-10-29:
Imagine not showing any screenshots and videos :roll_eyes:

aruznieto, 2023-10-29:
Yes. Month ago works

—

Andrés Ruz Nieto

GameTec-live, 2023-10-29:
you sure you didnt switch up the cards, because my mifare classic 4k works... proxmark 3 `hf search` returns mfc 4k?

aruznieto, 2023-10-29:
> Imagine not showing any screenshots and videos :roll_eyes: 

I am not at home now

aruznieto, 2023-10-29:
> you sure you didnt switch up the cards, because my mifare classic 4k works... proxmark 3 `hf search` returns mfc 4k?

Is the university card... impossible switch

Foxushka, 2023-10-29:
Do you remember ATQA/SAK?

aruznieto, 2023-10-29:
> Do you remember ATQA/SAK?

SAK: 38
ATQA: 00 02

GameTec-live, 2023-10-29:
> > Do you remember ATQA/SAK?
> 
> SAK: 38 ATQA: 00 02

my 4k got a SAK of 18, atqa is also 00 02 🤔 
Afaik 38 can be both mfc 4k and desfire, maybe thats why the app says tech other?

PR #453 (2024-02-17): Don't rely on SAK/ATQA to determine the number of blocks
Should close #367
closed 2024-02-17 by GameTec-live