GameTec-live/ChameleonUltraGUI #369: No device listed despite allegedly connected

Still34, 2023-10-30:
**Describe the bug**

[image]
ChameleonUltraGUI shows an empty list of connected device on the Windows client, despite the ChameleonUltra showing the loading bar animation. COM5 appears to be actively in use when the application is open, suggesting the port is opened and in use by the application. 

The CLI version works fine (compiled locally according to instructions) and can connect to the hardware just fine.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open the application
2. Connect the device
3. Click refresh
4. No application listed

**Expected behavior**
The device to show up in the application.

**Version info:**
 - OS: Windows 11
 - App build number: 686
 - Connection type: USB
 - Chameleon firmware commit: `gd77e166`

**Also required**:

- [ v ] - I've upgraded app to latest app version

- [ v ] - I've upgraded Chameleon firmware to latest version

- [ v ] - I've searched and verified that there is no already created issues

GameTec-live, 2023-10-30:
very strange issue... Is nothing else using the COM Port? (eg the CLI)

Still34, 2023-10-30:
Yes, nothing else should be using the port. Only either the CLI or the GUI is in use at a time.

GameTec-live, 2023-10-30:
mind enabling production logging in the debug settings and sending a log?

Still34, 2023-10-30:
```
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM1
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connected to COM1
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:104)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Product: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM5
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connected to COM5
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:104)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Product: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM1
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connected to COM1
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:104)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Product: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM5
```

GameTec-live, 2023-10-30:
your COM ports somehow dont return any manufacturer or Product information? are you running a custom firmware?

Still34, 2023-10-30:
Nope. The latest available 2.0.0 firmware `- Chameleon Ultra, Version: v2.0 (v2.0.0-110-gd77e166)`

Still34, 2023-10-30:
For additional context, the Ultra connects just fine and shows up on the GUI on my Linux laptop and iOS device.

GameTec-live, 2023-10-30:
thats very wierd, because it literlly is the indentical code... (And it works fine for me)

Maybe a driver issue?

Foxushka, 2023-10-30:
Chameleon is COM1 or COM5 in your case? PM3 client also interferes with our serial library. Try closing all software that can access serial ports

Foxushka, 2023-10-30:
If you see loading animation on chameleon before selecting chameleon in menu, then there is other software that using it. On main screen we just enumerate them, without connecting

Still34, 2023-10-30:
>Chameleon is COM1 or COM5 in your case? PM3 client also interferes with our serial library. Try closing all software that can access serial ports

Chameleon should be on COM5 - not entirely sure what's on COM1. 

> If you see loading animation on chameleon before selecting chameleon in menu, then there is other software that using it. On main screen we just enumerate them, without connecting

The animation starts as soon as I open the application, and vice versa when I close it.

Still34, 2023-11-03:
I just tried it on yet another device - same behavior.

Foxushka, 2023-11-03:
Can you try on `test-without-exception-handling` branch? Will it throw any errors?

Still34, 2023-11-03:
I'll test it later tonight. This laptop doesn't have flutter configured yet.

Foxushka, 2023-11-03:
You can grab compiled version from https://github.com/GameTec-live/ChameleonUltraGUI/actions/runs/6741642455

Still34, 2023-11-03:
Same issue unfortunately.

Foxushka, 2023-11-03:
Issue still will be there, something in logs?

Still34, 2023-11-03:
Roughly the same logs yeah
```
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM4
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM4
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connected to COM4
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:104)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Product: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to COM4
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connected to COM4
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:104)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Product: null
```

foXaCe, 2023-12-04:
hello i have same problem on arch linux

GameTec-live, 2023-12-04:
@foXaCe Mind providing some more info? kernel version, app version, debug log output, etc?

foXaCe, 2023-12-04:
@GameTec-live  yes sorry

Operating System: XeroLinux 
KDE Plasma Version: 5.27.9
KDE Frameworks Version: 5.112.0
Qt Version: 5.15.11
Kernel Version: 6.6.3-zen1-1-zen (64-bit)
Graphics Platform: X11
Processors: 16 × AMD Ryzen 7 3700X 8-Core Processor
Memory: 15.5 Gio of RAM
Graphics Processor: AMD Radeon RX 6600 XT

` 
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyS0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyACM0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyS0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyACM0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyS0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyACM0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyS0
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:88)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:54)
🐛 Connecting to /dev/ttyACM0`

[image]

GameTec-live, 2023-12-04:
@foXaCe Seems all up to date, only "wierd" thing would be the zen kernel, but that shouldnt change any functionality...
Do you have ModemManager installed? That likes to cause problems. Also; make sure your user has permission to access /dev/ttyS0 (uucp i think is the group on arch based systems)

foXaCe, 2023-12-04:
how to ?

foXaCe, 2023-12-04:
sudo gpasswd -a stephane uucp

do not work

foXaCe, 2023-12-04:
ok work after reboot

thank you

GameTec-live, 2023-12-04:
> ok work after reboot
> 
> thank you

so its fixed for you? great 👍

foXaCe, 2023-12-04:
yes thank you

493505110, 2024-10-02:
I have same problem when I use android device to connect it with BLE,but MTools BLE is working.

tri4ng1e, 2024-11-02:
I also has this issue on Windows. On Android, I can connect to the device with both MTools and ChameleonUltraGUI. On the Windows PC, I can connect with the console client, but ChameleonUltraGUI doesn't work.

The logs look the same, indicating that manufacturer and product are null:
```
🐛 Connecting to COM12
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:101)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:53)
🐛 Connected to COM12
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:102)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:53)
🐛 Manufacturer: null
#0   NativeSerial.connectDevice (package:chameleonultragui/connector/serial_native.dart:103)
#1   NativeSerial.availableChameleons (package:chameleonultragui/connector/serial_native.dart:53)
🐛 Product: null
```

Hence, `if (checkPort!.manufacturer == "Proxgrind")` line from `lib/connector/serial_native.dart` discards the device as a non-Chameleon device.

However, USB Device Tree Viewer shows that the device, in fact, has both manufacturer and product fields, which should pass the checks in `serial_native.dart`:
```
Manufacturer String      : Proxgrind
Product String           : ChameleonUltra: hw_v1, fw_v512
```

I don't know much about Flutter and Dart, but if my understanding is correct, ChameleonUltraGUI uses `flutter_libserialport` library, which is a wrapper around Dart's `libserialport` library, which is a wrapper around C library. On the Github page for Dart library, I found a somewhat recent issue related to productID, vendorId, manufacturer and description fields being null on Windows.
https://github.com/jpnurmi/libserialport.dart/issues/85

GameTec-live, 2024-11-02:
So it looks like we know the issue?