GameTec-live/ChameleonUltraGUI #372: Cant select NTAG213/215/216 when setting up slot

p0ns, 2023-11-14:
**Describe the bug**
When I go into the Slot Manager, edit the current HF card and try to select any of the NTAG options, it reverts back to Unknown.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Slot Manager
2. Click on any of the slots settings button
3. In the Edit Slot Data display, select NTAG213/215/216
4. It reverts to Unknown

**Expected behavior**
NTAG options should be able to be selected.

**Screenshots**
[image]

[image]

**Version info:**
 - OS: MacOS 14.0
 - App build number: 1.0.377 build 377
 - Connection type: USB
 - Chameleon firmware commit: g4d670f0

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [x] - I've searched and verified that there is no already created issues

GameTec-live, 2023-11-14:
hm, yes, this indeed seems to be a bug, but as firmware doesnt yet have full support for ntag anyways it isnt high priority for now...

closed 2023-11-15 by Foxushka

GameTec-live, 2023-11-15:
> Done on a purpose, firmware doesn't support it

ik fw doesnt support reading, but emulating too?