GameTec-live/ChameleonUltraGUI #393: Use ONLY custom dictionary

Stepzor11, 2023-12-12:
Every time it tries to read a tag, it cycles through its whole dictionary and the custom one (if chosen).
It would save a lot of time if I can chose to use ONLY a custom dictionary, without being forced to use also the system one.

GameTec-live, 2023-12-12:
afaik if you select a custom dictionary it only uses the keys in said custom dictionary

Stepzor11, 2023-12-12:
Nope.

I've tested and I can confirm that, as the text says, it is an "additional key dictionary" and it is used combined with the system one 😁

GameTec-live, 2023-12-12:
[image]
code says otherwise

GameTec-live, 2023-12-12:
maybe its performing key recovery

Stepzor11, 2023-12-12:
> maybe its performing key recovery

Nope, key recovery is after the dictionary attack (or at least should be after, with the specific button) 😁

GameTec-live, 2023-12-12:
> > maybe its performing key recovery
> 
> Nope, key recovery is after the dictionary attack (or at least should be after, with its specific button) 😁

yeah, right lol

GameTec-live, 2023-12-12:
hm, indeed... Very wierd...

closed 2023-12-24 by GameTec-live