GameTec-live/ChameleonUltraGUI #400: Bug when downloading cards from chameleon

VarChar42, 2023-12-17:
**Describe the bug**
Cards downloaded from the chameleon aren't saved correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Downlaod Card from chameleon (in my case Mifare Classic 1K EV1)
2. Save card as json
3. Examine json
4. Everything is in the first sector.

**Expected behavior**
Data gets stored into its corresponding sector.

**Screenshots**
Unexpected:
[image]
Expected
[image]

**Version info:**
 - OS: Windows 11
 - App build number: 765
 - Connection type: USB

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [x] - I've searched and verified that there is no already created issues 

**Additional context**
Despite this loading a card back onto the chameleon seems to work fine.

When viewing a downloaded card in the Hex Viewer (hex-editing branch) all data is also shown in Sector 0.

GameTec-live, 2023-12-21:
Ive come to the conclusion that its a error in the rebuildCardSaveFromSlot function... (So not only json would be corrupt, but everything else too...)
Now the question is where...
https://github.com/GameTec-live/ChameleonUltraGUI/blob/edbd1343514ae69aaf08b946ff037edcd7714ab7/chameleonultragui/lib/gui/menu/slot_export.dart#L40C4-L40C4

VarChar42, 2023-12-21:
Data gets returned like this from mf1GetEmulatorBlock. Maybe theres a problem with the chameleon firmeware.
[image]

GameTec-live, 2023-12-21:
> Data gets returned like this from mf1GetEmulatorBlock. Maybe theres a problem with the chameleon firmeware.
> [image]
> 

that'd be weird, but a possibility... Needs further investigation for sure...

commit cc7e2a22 (2023-12-23): Proper fix saving dump from Chameleon (closes #400)
closed 2023-12-23 by Foxushka