GameTec-live/ChameleonUltraGUI #410: Unable to run dark side attack

sarah1331sarah, 2023-12-26:
Advise please 

If your unable to run a dark side attack how can I run another type of attack please

GameTec-live, 2023-12-26:
Okay, some more information please... What are you trying to do? Did you skip the dictionary check? What app version are you running on? What OS? Steps to reproduce? ...

sarah1331sarah, 2023-12-26:
Sorry fw 2.0 v2.0.0-128 gfefcde5 

version 1.0.0 build 811

no default keys 
Recover keys 

and resulted in 
[image]

if I check keys from directory I get no keys all X

sarah1331sarah, 2023-12-26:
On another card (same campsite) I get sector 7 key b with all F’s but left to recover keys never recovers any after 12hours

GameTec-live, 2023-12-26:
one of your cards probably isnt voulnerable to darkside then... (the other one taking an eternety, idk) why skip default keys 🤷

sarah1331sarah, 2023-12-26:
I have no keys saved and default keys don’t hit either 

is it possible to run nested attack? When home I’ll try with offline nested with the card reader

GameTec-live, 2023-12-26:
for nested you need atleast a few keys

GameTec-live, 2023-12-26:
try using the larger pm3 dictionary, you can download it at their github:
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/dictionaries/mfc_default_keys.dic

sarah1331sarah, 2023-12-26:
sorry to sound thick how do i import the keys

GameTec-live, 2023-12-26:
saved cards -> dictionaries -> + -> select the downloaded file

sarah1331sarah, 2023-12-26:
With my mini I used to be able to sniff a key from the reader is that still possible

GameTec-live, 2023-12-26:
not yet, still very early stages... you are able to do mfkey32 though... Please reference the Chameleon Ultra docs for more info on that...

closed 2023-12-26 by GameTec-live