GameTec-live/ChameleonUltraGUI #422: Cannot save card

lordrak007, 2024-01-02:
**Describe the bug**
Can not save readed card. 

**To Reproduce**
Steps to reproduce the behavior:
1. Read whole card
2. Click save, enter name
3. After click to save button nothing happens

If i use save only UID, uid is saved successfully. 

**Expected behavior**
Saved whole card

**Screenshots**
Nan

**Version info:**
 - OS: Android 13
 - App build number: 392
 - Connection type: BLE
 - Chameleon firmware commit: dont know, current

**Also required**:

- [ ] - I've upgraded app to latest app version

- [ ] - I've upgraded Chameleon firmware to latest version

- [ ] - I've searched and verified that there is no already created issues

GameTec-live, 2024-01-02:
it says on the homepage...

GameTec-live, 2024-01-03:
also; card saving works fine for me...

lordrak007, 2024-01-03:
What it says on the homepage?
A have all up to date, and finally found FW version (edited in request).
Still, i cant save the card. When i try to read card, then click to save, enter name and nothing happens...

GameTec-live, 2024-01-03:
weird, as said, saving works fine for me...
what card? mfc?
Also, what did you try to name it?

lordrak007, 2024-01-03:
Yes, i am trying nfc Miface Clasic 1k. On windows working, on android nothing. Are there some lpgs which i can check or collect?

GameTec-live, 2024-01-03:
yeah, you can try and perform saving a card with logging enabled... (a screenshot from the "read card" page, before you click on "save" cant hurt either)

To enable production logging:
1. Go to settings
2. Activate Debug mode
3. Go to Debug settings
4. Enable production logging

Then perform your actions, copy logs and dont forget to disable it again:
1. Return to Debug Settings
2. Click on "Copy logs to clipboard"
3. Paste them in github (please format them as a codeblock or upload a textfile)
4. Disable production logging
5. Press "Clear logs" (As leaving the logs in the "database" increases apps storage requirements and startup time)
6. (recommended) Disable Debug settings under Settings page

What does this option do? it writes all debug messages to disk (incase the app crashes) and allows you to copy them. This does mean the app will be a lot slower with prod logging enabled though.

lordrak007, 2024-01-04:
I did it, here is problem with test card.
[log.txt](https://github.com/GameTec-live/ChameleonUltraGUI/files/13827443/log.txt)

GameTec-live, 2024-01-04:
> I did it, here is problem with test card.
> [log.txt](https://github.com/GameTec-live/ChameleonUltraGUI/files/13827443/log.txt)
> 

ill look at it later, thx

lordrak007, 2024-02-02:
No, Czech.

PR #452 (2024-02-04): fix: fixes cards not saving on other languages
closes: #422
closed 2024-02-04 by Foxushka