GameTec-live/ChameleonUltraGUI #473: Firmware update fails

mindbreaker2, 2024-04-06:
When I try to update the firmware using the iOS cu gui app the download begins, then installation get stuck because the cu seems to shut off.

**To Reproduce**
Steps to reproduce the behavior:
1: connect to the cu 
2: use the look for update button

It used to work, I've updated the firmware multiple times but not working anymore

**Version info:**
 iOS 17.2.1
Cu gui up to date 
Cu firmware 2.0.0-131

GameTec-live, 2024-04-06:
Yes, its a known issue... Apperenlty BLE DFU is broken. For now just update via a PC and USB please until theres time to fix it...

mindbreaker2, 2024-04-06:
Yep, works like a charm using usb connection.

Hadn't seen any issue about it so I wanted to let everyone know. 

Thanks

PR #477 (2024-04-23): fix: BLE DFU flashing and update privacy info
Closes #473
closed 2024-04-23 by GameTec-live