GameTec-live/ChameleonUltraGUI #480: Adding BLE support on desktop app

stuiterveer, 2024-05-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Currently, the desktop app only supports a direct USB connection. Still, Bluetooth Low Energy is available to use on computers with a (relatively) recent bluetooth interface will allow setting up a BLE connection and interfacing with it. One notable example for Windows, MacOS and Linux is [Bleak](https://github.com/hbldh/bleak) which gives some easier to manage Python calls for Bluetooth without having to double-check the platform the user is on.

Main reason for requesting this feature is because my main driver is mainline Linux instead of Android/iOS and I would still like to use the desktop app in a similar method compared to those smartphones.

GameTec-live, 2024-05-21:
Did you look at the repo? This isnt a python app... Also; maintaining Bluetooth on android and iOS can be done with one library and is already painful enough, having another Library i need to worry about, when most PCs dont support BLE, isnt worth it imo...

GameTec-live, 2024-05-21:
Your free to implement it yourself and submit a PR though!

stuiterveer, 2024-05-21:
Bleak in this case was meant as an example. I haven't been able to dig into the code for the GUI yet so I'm currently requesting it from a place of ignorance where I don't know how much effort it will take to get done and I don't know how easy it is to get done in the likes of Flutter or whatever is used for the BLE part. [This](https://pub.dev/packages/universal_ble) suggests is (relatively) trivial though.

GameTec-live, 2024-05-21:
Just because a package exists that provides a API for talking with bluetooth doesnt mean that its trivial...

JackyYuenDacai, 2025-10-04:
universal_ble seems have wider platform supports