GameTec-live/ChameleonUltraGUI #489: new logo?

leandroebner, 2024-07-07:
hey. This is super low level priority but what is the app icon actually displaying?
I thought its a chameleon but I cant really see it. It always just looks like a broken image file. 
That would also improve the overall appearance in the app-store (play-store) and maybe attract more people. 
Just a thought for the far future, Id like to help out as well if a design is needed!

Cheers, thanks for that great project!

GameTec-live, 2024-07-07:
Yeah, its supposed to be the Chameleon in some weird rainbiw particle thing. I know, its horrible, but im no graphics designer, so its just something i chucked together in blender when i was bored.
If you know how to do graphics, feel free to try and make a logo, if its great, maybe itll become the new logo ;)

leandroebner, 2024-07-07:
no worries!
perfect, Ill keep that in mind when Im doing other design work too :)

nasaboy, 2024-11-02:
Looking forward to your design of a new logo.

Saint-Ricky, 2024-12-20:
Not sure if you can use one or both of these but I thought I’d give it a go. 
[image]
[image]

GameTec-live, 2024-12-20:
@Saint-Ricky Dont want to be mean, but they do look to close to the gold printing on the front side of the CU for my taste. The app should still have its own identity while drawing a connection to the CU,.

Saint-Ricky, 2024-12-20:
@GameTec-live Yeah I wasn’t sure if that would be an issue or not. That was what I was going for at the time. I also threw this one together which is more abstractly a chameleon but with some radio signal coming off. Let me know what you think. I’m definitely open to suggestions and input. 
[image]

GameTec-live, 2025-01-05:
Thanks to the extremely talented @Saint-Ricky theres work on a new CUGUI logo. Please tell me which direction you like best or, if nothing fits, whats wrong with the existing ones / what youd like to see.

[image]
[image]
[image]
[image]
[image]

duchenpaul, 2025-02-22:
> @GameTec-live Yeah I wasn’t sure if that would be an issue or not. That was what I was going for at the time. I also threw this one together which is more abstractly a chameleon but with some radio [...]

Actually I like this design! Though it looks like AI artwork, but love it at the first glance