GameTec-live/ChameleonUltraGUI #510: small key inconsistentency

mitmarcus, 2024-08-14:
[image]

[image]

Key type is not the same

GameTec-live, 2024-08-15:
Because theres no mfc1k ev1, atleast there no need to as a ev1 is just a 1k with extra sectors, using a 2k has more sectors so we can fit them and the rest just gets ignored

closed 2024-08-15 by GameTec-live

mitmarcus, 2024-08-15:
> Because theres no mfc1k ev1, atleast there no need to as a ev1 is just a 1k with extra sectors, using a 2k has more sectors so we can fit them and the rest just gets ignored

ah thank you! that explains it