GameTec-live/ChameleonUltraGUI #526: LF tag, 8-byte ID support

matteoscrugli, 2024-11-05:
Is there any way to emulate an LF tag with an ID of 8 bytes instead of 5? I can only succeed with the Flipper Zero, but not with the Chameleon Ultra.

Thank you very much for the great work

GameTec-live, 2024-11-06:
As youve found out, theres no way rn.
Stuff like tag support is a fw issue. Look at the fw docs to see what's supported and what can be supported.

closed 2024-11-06 by GameTec-live

matteoscrugli, 2024-11-06:
> As youve found out, theres no way rn. Stuff like tag support is a fw issue. Look at the fw docs to see what's supported and what can be supported.

According to the support tables, the firmware should support T5577 tags... In teory, T5577s allow an 8-byte ID. So I would say it's an issie on the software side, right?

GameTec-live, 2024-11-06:
[image]

No it doesnt.
It has to say support on all 3
This just means that it CAN be implemented

matteoscrugli, 2024-11-06:
> ![image](https://private-user-images.githubusercontent.com/66077766/383485234-9b57e2c4-df55-416b-9ce1-d630065142f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4ODY3MjcsIm5iZiI6MTczMDg4NjQyNywicGF0aCI6Ii82NjA3Nzc2Ni8zODM0ODUyMzQtOWI1N2UyYzQtZGY1NS00MTZiLTljZTEtZDYzMDA2NTE0MmY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA2VDA5NDcwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0MDIxMzYyNDNmY2E4ZGU3YWMwNDM4OTMxNGE1YWExNTY1YWVmYjA4YTI3YzQxZTc2YTk0YmY3MTgwNWY3NjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.PtsAEk6q63CVbdytBe_F2jeBncCxWPcWdxOHg4bsxBM)
> 
> No it doesnt. It has to say support in all 3 This just means that it CAN be implemented

I thank you for your answer and apologize if I am not understanding.

It's not clear to me at this point if you're saying it's firmware problem or an application problem. I understand, in contrast to your first message, that this is not a firmware problem. If the problem is at the application level I might try to take a look at the code. 

Do you have an idea what component you can get your hands on? Thank you very much again

GameTec-live, 2024-11-06:
It's a fw problem
It isnt implemented in the FIRMWARE Application
If it's listed in the table, the hw could support it.
If it doesn't say support in all 3, fw cant do it yet