GameTec-live/ChameleonUltraGUI #536: Unable to install deb package on debian 10

GrafOrlok, 2024-12-10:
**Describe the bug**
I'm unable ton install the deb package provide

**To Reproduce**
Steps to reproduce the behavior:
1. unzip the archive linux-debian.zip
2. `run sudo dpkg -i chameleonultragui_1.0.0_amd64.deb`
3. the error above occure :

dpkg-deb: error: archive 'chameleonultragui_1.0.0_amd64.deb' uses unknown compression method for member 'control.tar.zst'. Aborting. dpkg: error processing archive chameleonultragui_1.0.0_amd64.deb (--install): dpkg-deb --control subprocess returned error exit status 2 Errors were encountered during execution: chameleonultragui_1.0.0_amd64.deb

**Version info:**
 - OS: Debian 10

GameTec-live, 2024-12-10:
Stupid question, but isnt debian 10 like 2 OS versions behind?

closed 2024-12-10 by GameTec-live

GrafOrlok, 2024-12-10:
no problem, I plan to upgrade my debian install soon.