GameTec-live/ChameleonUltraGUI #540: ios 18.1.1 not found CU

dittohead, 2024-12-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to app
2. Click on refresh button and click any button on CU
3. Wait

**Expected behavior**
CU in list, able to connect

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version info:**
 - OS: ios 18.1.1
 - App build number: 1.0.452 appstore (f80487c #538)
 - Connection type: ble
 - Chameleon firmware commit: last master

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [ ] - I've searched and verified that there is no already created issues 

**Additional context**
Add any other context about the problem here.

GameTec-live, 2024-12-21:
Hm, works fine on my device. Make sure the CU is woken up when refreshing.

dittohead, 2024-12-22:
I charges CU and it start to detect, pretty weird

closed 2024-12-22 by dittohead