GameTec-live/ChameleonUltraGUI #570: Pairing BLE android 13 and 14 Xiaomi

Alberto0581, 2025-04-12:
Hello, I wanted to ask you about an issue I'm having with Bluetooth pairing with a Chameleon Ultra using the CU GUI app.
I've run several tests with different Android smartphones, with different versions.
Mi112T Android 13
Poco M5 Android 14
Redmi Note 8 Pro Android 10.
The problem I'm having is that once I pair the Chameleon Ultra's Bluetooth, it doesn't appear but stays paired. This is with CU GUI version 1.2 on the Mi12T and Poco smartphones. The Chameleon Ultra is working on Redmi Note 10.
I then installed the first 0.0.0 versions via APKPure, and it works on all the smartphones I've tested.
All versions work perfectly using a USB-C cable.
I've also been fiddling with some of the Mi12T's developer options, but they still don't work.
It also works on android 5.
Question: Could it be something due to the layers that Xiaomi has put in place? I also want to say that I have all the permissions granted on Xiaomi: precise location, contact sharing, autostart. Pairing, it pairs because it stays connected but doesn't appear again in the app via Bluetooth. Thanks, and sorry for the long text.

lazzman, 2025-05-26:
The same issue with vivo x200 pro as well.
```
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:72)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:101)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.connecting, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.connected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:72)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:101)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.connecting, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.connected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
[38;5;208m#0   BLESerial.connectSpecificInternal.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:151)[0m
[38;5;208m#1   _RootZone.runUnaryGuarded (dart:async/zone.dart:1778)[0m
[38;5;208m⚠️ ConnectionStateUpdate(deviceId: FB:BE:F7:1F:AB:80, connectionState: DeviceConnectionState.disconnected, failure: null)[0m
#0   BLESerial.availableDevices.<anonymous closure> (package:chameleonultragui/connector/serial_ble.dart:72)
#1   Timer._createTimer.<anonymous closure> (dart:async-patch/timer_patch.dart:18)
🐛 Found BLE devices: 1
#0   BLESerial.availableChameleons (package:chameleonultragui/connector/serial_ble.dart:101)
#1   <asynchronous suspension>
🐛 Found Chameleon Ultra!
```

Foxushka, 2025-06-29:
@lazzman @Alberto0581 

Will be very helpful if you can check most of versions and pin point each moment when it got broken (as we can't reproduce). For general versions all major are interesting (1.1, 1.0, 0.0 latest) https://apkpure.com/chameleon-ultra-gui/io.chameleon.ultra/versions
If doesn't work on 1.0, try finding latest 0.0.0 build where it was working, unfortunately matching builds to each commits isn't possible, but we can try to match on date. Dates should match commits on those files:
https://github.com/GameTec-live/ChameleonUltraGUI/commits/main/chameleonultragui/lib/connector/serial_ble.dart
https://github.com/GameTec-live/ChameleonUltraGUI/commits/main/chameleonultragui/pubspec.yaml

Ussrnm, 2025-07-04:
Hello! 
The same issue with xiaomi redmi note 9 pro, android 12. CU GUI v 1.2.0. With using a USB-C cable CU GUI work. Thank you.

Foxushka, 2025-07-04:
@Ussrnm if you can reproduce this issue, you should read my previous message and help us find commit where exactly BLE is broken. As I said previously, we can't reproduce this issue

Foxushka, 2025-07-15:
@msadlis can you try on 388? Downloads fine for me

msadlis, 2025-07-15:
@Foxushka 388 BLE not connecting

Foxushka, 2025-07-15:
@msadlis try those 
https://play.googleapis.com/download/playconsole/AOTCm0R_plrrNuuINAEQVCZOHPE9jGeEB82eHGxPIuAfAZUMSxMJdcp3q7L4cSXTuQKuBkVQpQ-x2zBoSKK8gu5xXhI4sVo1hUs-dgjRaN6HTbgHEDGVu_kq1Wh0FBKs15sdL_3suLYXK0miwpBOpFL93N7u76Nn9VALhMARzHtHxh_5WGKmfdPg9oDCw8gUKsssbrrJ
https://play.googleapis.com/download/playconsole/AOTCm0SJdjSHWEQeQQMsr95K-SEePSMwUogZI-0P5-FMNNUUKifo_3XMog8cehE9XI1D_t_Gzh3BIk4C3NfIoASiJsVOVpIx6Dc3sPpValLqNpwgNgom4GInvGqt1IGKhyW9hjqKiyz26mC-1eSGoWcAVHZSdeGImG7xWR-owIwmb954w2i1sUmlof26-o56If05icOO
https://play.googleapis.com/download/playconsole/AOTCm0TilNV4y46hatXK7-tKetaleKjfCpdIoJxotxvsBHjNffKiPLVJI-tp8dna166XlSmtavM-2Ce8VFWVQ21FO_ThqHhFGFr4hgJ0B04joKecBfK4alwZ5_N1QJSzx9iUbfqyN7eM13u9OwnHwVPCWMR_Uab7iA4Z-txS1Garocb8MaPSLtFKBh6Ew91TQpT5UK4

msadlis, 2025-07-15:
@Foxushka 225 BLE works, there is a connection

Foxushka, 2025-07-15:
@msadlis 
https://play.googleapis.com/download/playconsole/AOTCm0TWfkRjm8dpIW73Efp_MvxNjutNldTmUPuNpJTNV2biZfQoWeRhpVmzJ49tSvWRGi39JvhJ1Samejr97wfHH0FVcbKpCMTvIEmRWA8f_el_H2WJJdaGehPGWnQNYk-4MG9rDeWgpye_8LnnyVP5vi9HHyItUkw6ABld8dBMT5HY-7LYI3u8pggt6rN2gHHs9Hw
https://play.googleapis.com/download/playconsole/AOTCm0TKMCkxG0Eudt0qB4KK0h8TrQmK2ivmzL5qShKsNkUUBLnzf1Xw7pQeeAgVU7cHpXTouzdKHihQWVrYGr_KP3WzZP6Vhw0HyCK0-cbv4YO80ixpdqMoUPBfmhVVKaLIcEjFVGJAnGGmnQcfOCvJ4yNrOuSS1RWpUhZm2Ps7Mb5b8YCSmIB76ULNYqKVvTeWaMI

msadlis, 2025-07-15:
@Foxushka 241 BLE not connecting

Foxushka, 2025-07-15:
@msadlis https://play.googleapis.com/download/playconsole/AOTCm0SonhTfmtSTcwjmwCbNpf9Vp7hz6pD9DMCg79wrOnAmbamoL4YgwVbOsV3Qo1ZR7IY-IazKgBQg9lc2PQdZqDcfz_ByQRiULcHeEgguLKXE_F4Xun3UvXfYHZZtLR4hMCq8F6JFeRwuP6HIJFW9EDEhFCqfBv--12PWGSc7q3mzoFx2R2yhzEkoTjMchkLQ57I

msadlis, 2025-07-15:
@Foxushka 235 BLE works, there is a connection

msadlis, 2025-07-15:
@Foxushka 236 BLE works, there is a connection
237 BLE not connecting

Foxushka, 2025-07-15:
@msadlis https://github.com/GameTec-live/ChameleonUltraGUI/actions/runs/16296401408/artifacts/3536866020 try this

msadlis, 2025-07-15:
@Foxushka in this version BLE works, there is a connection

msadlis, 2025-07-15:
@Foxushka BLE works, there is a connection, thank you.

closed 2025-07-15 by Foxushka