GameTec-live/ChameleonUltraGUI #571: Incorrect import to "saved" drom device slot when enabled only HF on the slot

AMajor-C, 2025-04-14:
**Describe the bug**
Incorrect import to "saved" from device slot when enabled only HF on the slot.
It saves uid from any other existing slot.

**To Reproduce**
Steps to reproduce the behavior:
1. Save a hf card to slot where no lf card added
2. Enter it and disble the lf card at all by the switch.
3. Now try to export to saved card

**Expected behavior**
It saves uid from any other existing slot. 

**Version info:**
 - OS: W10
 - App build number: 1151
 - Connection type: [e.g. BLE]
 - Chameleon firmware commit: gde27b6

**Also required**:

- [ ] - I've upgraded app to latest app version

- [ ] - I've upgraded Chameleon firmware to latest version

- [ ] - I've searched and verified that there is no already created issues

GameTec-live, 2025-04-15:
Sorry, but i dont understand what you mean?
For me, saving a card from a slot works fine?

Foxushka, 2025-06-26:
Not reproducible, you can reopen with video if you can reproduce it on latest firmware + app versions