GameTec-live/ChameleonUltraGUI #660: app写卡页面显示异常

a2359765717, 2025-08-17:
**Describe the bug**
app写入卡片第三步显示异常

**Version info:**
 - OS: [Android 15]
 - App build number: [1486]
 - Connection type: [BLE]

[image]

**Also required**:

- [ *] - I've upgraded app to latest app version

- [ *] - I've upgraded Chameleon firmware to latest version

- [ *] - I've searched and verified that there is no already created issues 

**Additional context**
Add any other context about the problem here.

GameTec-live, 2025-08-17:
can you please file the issues in english?

a2359765717, 2025-08-17:
> 您能用英文提交问题吗？

I just noticed that someone has already submitted my question.

T55XX writing page error #659

closed 2025-08-19 by Foxushka