GameTec-live/ChameleonUltraGUI #679: Static encrypted nonces attack seems to not be able to recover keys by only using the backdoor key

GameTec-live, 2025-08-30:
**Describe the bug**
I'm testing the new implemented static encrypted nonces attack to recover keys on a fudan card.
Chameleon is able to find the keys if I run first the dictionary attack with a known key, but it fails if I skip the dictionary attack (so it should use the backdoor key for the attack as the first key).
The card has 2 "protected" sectors, 0 and 1, with "unknown keys", sectors 2 to 15 have ffffffffffff keys.

**To Reproduce**
Steps to reproduce the behavior:
1. Using chameleon ultra gui for windows, go to read tag, click on read
2. Click on skip keys in predefined dictionary, click on check keys in dictionary, click on recover keys
3. wheel starts rotating for key A0, and stays there, progress bar advances and once it reaches 100%, after about 1h and a half it resets to 0% and starts again
4. no key is recovered

**Expected behavior**
Using the backdoor the behaviour should be similar to that of using a known key of another sector found with the dictionary attack; with the found key first (ffffffffffff) the chameleon is able to recover the "unknown keys" in a couple of minutes.

**Screenshots**
This is a hf mf info from the proxmark
[image]

**Version info:**
 - OS: Windows 11
 - App build number: chameleon ultra gui release 1.3
 - Connection type: wired
 - Chameleon firmware commit: 09870c3

**Also required**:

- [x] - I've upgraded app to latest app version
- [X] - I've upgraded Chameleon firmware to latest version
- [X] - I've searched and verified that there is no already created issues 

Logs:
log from a successful run (dictionary attack run first)
[log-ok.txt](https://github.com/user-attachments/files/22058754/log-ok.txt)

Log from an unsuccessful run (no dictionary attack run first)
[log-not-ok.txt](https://github.com/user-attachments/files/22058755/log-not-ok.txt)

@82ghost82 this issue has been moved to the gui repo

Foxushka, 2025-08-30:
@82ghost82 can proxmark3 recover keys via `script run fm11rf08s_recovery.py`? I have only one 0490 from bambulab filament spool and it throws [#] AcquireStaticEncryptedNonces: Auth1 error when I try to run it via proxmark3

82ghost82, 2025-08-30:
yes, no issue with the proxmark, keys are found in 16 seconds with command `script run fm11rf08s_recovery.py`

Foxushka, 2025-08-30:
@82ghost82 
I've pushed potential fix, should build at https://github.com/GameTec-live/ChameleonUltraGUI/actions/runs/17346467613 in ~10 minutes or you can build app by yourself. Can you test it?

82ghost82, 2025-08-30:
it seems quite fixed, now I have 2 spinning wheels instead of one (a and b), it found the 2 "unknown" keys in sectors 0 and 1 (quite at the same time), next progress bar was resetted to 0% but it seems it's not able to find key ffffffffffff with this attack.
it's running for 15 minutes now, I think it should be enough to find the f key.
[image]

Do you need logs?

Foxushka, 2025-08-30:
Progress bar should fully be gone and jump to other sector, otherwise keys might be found even in last batch (maybe they are sorted from 0 to F's)

82ghost82, 2025-08-30:
it's ok, this can be closed @GameTec-live it's fixed, f key was found too. Thank you again @Foxushka for this fast fix!