GameTec-live/ChameleonUltraGUI #695: Mifare static encrypted key recovery runs in an infinite loop (>3 hrs)

Leka1B, 2025-09-03:
**Describe the bug**
The new update finally let me try to recover the missing keys for my card with static PRNG and no backdoor.
My longest key recovery attempt with a static encrypted nonce card took around 3 hours, and failed to recover even a single key.

**To Reproduce**
Steps to reproduce the behavior:
1. Read the card
2. See "PRNG type: Static" and "Backdoor: No"
3. Check keys from a dictionary and get a couple hits
4. Click "Recover keys" to get the rest
5. See a single spinner and a slowly-filling progress bar
6. Wait >40 min for it to fill
7. The bar resets to the beginning
8. Repeat

**Screenshots**
[image]

**Version info:**
 - OS: [Android 14]
 - App build number: [519] - Also tried on the latest dev build from commit 0fa89dd.
 - Connection type: [BLE]
 - Chameleon firmware commit: [g09870c3]

**Also required**:

- [x] - I've upgraded app to latest app version

- [x] - I've upgraded Chameleon firmware to latest version

- [x] - I've searched and verified that there is no already created issues 

**Additional context**
Could not get the logs for this, they were too long to copy (I tried to capture the whole cycle of the progress bar). (see issue [694](https://github.com/GameTec-live/ChameleonUltraGUI/issues/694))

Foxushka, 2025-09-03:
There is known issue with 0x009080A2 nonce ones, was confirmed with @GameTec-live. I've tried guessing required distance based on known key, but no luck
https://github.com/RfidResearchGroup/proxmark3/issues/2938

Can you run hf mf nested -a -k <key> --blk 0 --tblk 0 --tb via **CLI** and post line similar to: Executing ./staticnested 700875338 97 18874693 3840288840 18874693 3103511955 there?

Foxushka, 2025-09-03:
[keys.txt](https://github.com/user-attachments/files/22126631/keys.txt)  try those

Leka1B, 2025-09-03:
Those are wrong too

commit f70c374b (2025-09-14): feat(tool): Dictionary download and T55XX password cleaner (#702)

* fix: Artifacts priority for version check

* refactor: Separate definitions from bridge

* feat: Add keyLength to dictionary

* fix: Limit nested recovery tries to 1 for static cards

Closes #695, proper fix might be implemented after working method

* feat: Show recovery state to user

Closes #678

* fix: Localizations

* fix: Improve Mifare Classic EV1 recovery

And small refactor of checkmarks

* feat(tool): Download external dictionaries

Closes #647

* fix: Fetch from actions

* fix: Fix T55XX writing for blank tags + mi [... cut]
closed 2025-09-14 by Foxushka